GRIDAPPSD/gridappsd-forum #85: Installation on Windows

Lakes-lead, 2021-01-16:
Hello is there a guide on how to install GridAppsD in windows ?

Thank you

AAndersn, 2021-01-18:
Hello,

For application development, we recommend using the windows subsystem for linux. Complete instructions are available at 

https://github.com/GRIDAPPSD/gridappsd-hackathon/blob/main/Lesson%201.0.%20Installation%20Instructions.pdf

If you don't plan on using the API training notebooks, you can skip steps 1.4 through 1.7. 

Also, be sure to reach out for an invitation to the monthly app developers meeting.

Lakes-lead, 2021-02-11:
Hello Alexander,

Thank you for response.

Does the OpenDSS simulator now works with the GridAppsD platform or just the Gridlabsd for now . 

Thank you 

Best regards,

AAndersn, 2021-02-11:
Only GridLab-D currently. OpenDSS simulation will be added in a future release.

By the way, check out the new discussion board we created a few days ago. It is the perfect place to ask these type of questions and anything you need help / support / advice on.

https://github.com/GRIDAPPSD/gridappsd-forum/discussions

closed 2021-03-11 by poorva1209