GRIDAPPSD/gridappsd-forum #86: Query CIM model from Python

MaximeBaudette, 2021-01-25:
I'm trying to fetch models from GridApps D and would like the pull the full CIM XML to use further. 
I tried the following:
   ```
 message = {
        "requestType": "QUERY_MODEL",
        "modelId":model_mrid,
        "resultFormat": "XML"
    }

```
And got the following: 

  ```
  {'error': 'Invalid json returned',
     'header': {'content-length': '52460',
      'expires': '0',
      'destination': '/temp-queue/response.20210122Jan2428',
      'GOSS_SUBJECT': 'system',
      'subscription': '789543',
      'priority': '4',
      'transformation': 'TEXT',
      'GOSS_HAS_SUBJECT': 'true',
      'message-id': 'ID:5a1570326af6-37457-1611266567218-4:9:-1:1:25',
      'persistent': 'true',
      'timestamp': '1611354268850'},
     'message': '{"data":<?xml version="1.0"  
```

poorva1209, 2021-02-05:
@MaximeBaudette The gridappsd-python library is incorrectly tagging the response as an error because it is expecting the response to always be a valid JSON object. 
The CIM XML you are expecting is inside 'message' under 'data' key. You can use that. 
Meanwhile, I will make a pull request to untag this as an error and return valid XML.

MaximeBaudette, 2021-02-10:
@poorva1209 
Thanks! 
I have one more questions. The CIM data I extracted from the "data" section seems to be missing a bunch of "resources" as most IDs do not point to anything within the file.

poorva1209, 2021-02-10:
Let me check that and I will get back to you.

MaximeBaudette, 2021-02-10:
@poorva1209 for info, I'm looking at the IEEE13 model

poorva1209, 2021-02-11:
@MaximeBaudette  Can you paste a resource number here that you are not finding in the file.

MaximeBaudette, 2021-02-11:
@poorva1209 there's basically none of the resources in the file.
_17E0320F-9A09-4C78-8511-7DFF5F56E330 is one example

poorva1209, 2021-02-15:
@tdtalbot  Here is the issue regarding query model API not returning all the objects.

poorva1209, 2021-03-11:
> @MaximeBaudette The gridappsd-python library is incorrectly tagging the response as an error because it is expecting the response to always be a valid JSON object.
> The CIM XML you are expecting is [...]

@MaximeBaudette The API has been updated to not label XML response as an error. So with the latest develop tag, you should get a valid XML as the response. We are still working on making sure all the resources are present in the QUERY_MODEL response.

MaximeBaudette, 2021-03-24:
I updated to 2021.02.0 release and can easily fetch the xml model (and write it locally). 
Thanks for the update!

MaximeBaudette, 2021-03-24:
@poorva1209 Any idea as to why I'm not getting all the resources in the CIM file ? Should I update my request, or request additional model and merge them ?

poorva1209, 2021-03-24:
@MaximeBaudette We are working on making sure all the resources are included in the CIM file. Currently, this issue is in progress and we will update the status here as soon as it is merged in the develop branch. You don't need to update the request. Adding @tdtalbot.

MaximeBaudette, 2021-03-24:
Ok great, thanks! Will await update. 
In the meantime, is there anyway you could email me a full CIM file ? or do you have the issue to generate such file also in the background (not only in the API call)? 
I'm trying to get things going reading CIM files and I'm struggling to find any fully working CIM file for IEEE13 on the net.

poorva1209, 2021-03-24:
Please try this API call mentioned here: https://gridappsd.readthedocs.io/en/latest/using_gridappsd/index.html#request-cim-dictionary-file with the model mrid for IEEE 13 node model. 

Request topic: goss.gridappsd.process.request.config
Request: {
  "configurationType":"CIM Dictionary",
  "parameters":{"model_id":"model_mrid"}
 }
Response: CIM Dictionary with the full model

Let me know if this works for you or if you face any issues.

MaximeBaudette, 2021-03-30:
Hi Poorva, 
I tried this query, but I couldn't figure out a way to get a CIM XML file in response. It seems to only return a JSON.

poorva1209, 2021-04-29:
@MaximeBaudette Updated have been made to develop tag that would return proper QUERY_MODEL response in XML with all the resources. Please check and let me know if you see any issue.

MaximeBaudette, 2021-04-29:
Excellent, I'll test this out !

MaximeBaudette, 2021-04-29:
@poorva1209 
Hi Poorva, 
I'm trying to switch to the develop branch of the Gridapps container, and tries to change my gridapps python to develop too, but it won't work... any idea as to why?

poorva1209, 2021-04-30:
@MaximeBaudette Are you getting any error?

MaximeBaudette, 2021-04-30:
@poorva1209 
For some reason I have a lot of failure to run `./run.sh -t develop`  to start gridapps, buit when I manage to get it up and running and I go to my python notebook and execute the command to instantiate gridAPPSD I get the following: 

[image]

poorva1209, 2021-04-30:
@MaximeBaudette I have seen this error with the latest version of stomp.py. Can you do 'pip list | grep stomp' and make sure the version is 6.0.0. If not install that version and try again.

MaximeBaudette, 2021-04-30:
Thanks a lot ! now I can run the python API.

MaximeBaudette, 2021-07-13:
@poorva1209 
I have a question about the model I retrieve through the API call. 
CIM models I found on one of your repositories have connectivitynodes for each ACLineSegment, from which I can extract the terminal and build a network graph. 
For some reason, when I apply the same script to build network graphs from the models extracted from Gridapps through the API call I don't get any connectivity node in the ACLineSegments, am I missing something ?

MaximeBaudette, 2021-07-15:
I worked on the following query
```
# IEEE 13 Node used for all example queries
#model_mrid = "_49AD8E07-3BF9-A4E2-CB8F-C3722F837B62" 
model_mrid = "_5B816B93-7A5F-B64C-8460-47C17D6E4B0F"
# Fetch Model from GridApps
message = {
    "requestType": "QUERY_MODEL",
    "modelId":model_mrid,
    "resultFormat": "XML"
}

topic = t.REQUEST_POWERGRID_DATA

response = gapps.get_response(topic, message)
# Write model to file
from pathlib import Path

path_to_repo = Path('..').resolve()
path_to_CIM = path_to_repo / 'data' / 'CIM_files'
path_to_IEEE13 = path_to_CIM /'IEEE13_gridapps.xml'

xml_file = open(str(path_to_IEEE13.absolute()),'w')
xml_file.write(response)
xml_file.close()
```

AAndersn, 2021-07-16:
@MaximeBaudette , Please try this query message:

```
message = {
    "requestType": "QUERY_OBJECT_DICT",
    "resultFormat": "JSON",
    "modelId": "_5B816B93-7A5F-B64C-8460-47C17D6E4B0F",
    "objectType": "ConnectivityNode"
}
```

Thanks,

Alexander

MaximeBaudette, 2021-07-19:
Hi Alexander

I’m interested in the full CIM XML file rather than just a list of devices.

Best regards,

AAndersn, 2021-07-19:
@MaximeBaudette ,

I don't think the GridAPPS-D API has a call to return the entire XML model. You could clone it directly from GitHub (https://github.com/GRIDAPPSD/Powergrid-Models/tree/develop/platform/cimxml) or you can query for the entire model as a python dictionary:

```
topic = "goss.gridappsd.process.request.config"

message = {
    "configurationType": "CIM Dictionary",
    "parameters":{"model_id": "_4F76A5F9-271D-9EB8-5E31-AA362D86F2C3"}
}

gapps.get_response(topic, message, timeout = 30)
```

@poorva1209  @craig8  Is there a query to return the original CIM XML that I am not aware of?

MaximeBaudette, 2021-07-20:
Why doesn't the power grid model API return the power grid model ?
Maxime

poorva1209, 2021-07-20:
Hi @MaximeBaudette , I executed this query and I can see connectivity node objects in the response. Is there anything specific I should be looking for that is missing? 

{
        "requestType": "QUERY_MODEL",
        "modelId": "_49AD8E07-3BF9-A4E2-CB8F-C3722F837B62",
        "resultFormat": "XML"
}

MaximeBaudette, 2021-07-20:
Hi @poorva1209 
So I had a closer look at the list that @AAndersn provided on github. 
Here is the model I'm referencing to
https://github.com/GRIDAPPSD/Powergrid-Models/blob/develop/platform/cimxml/IEEE13.xml

When querying the API with mRID _5B816B93-7A5F-B64C-8460-47C17D6E4B0F
I receive the file with Connectivity Nodes, but somehow they are not linked to anything, upon further investigation, the file is missing all the terminals that provide the linkage between topo / Connect nodes and equipment such as ACLineSegment. 

I tried mRID _49AD8E07-3BF9-A4E2-CB8F-C3722F837B62 as well and also received a file without any Terminal classes

poorva1209, 2021-07-20:
@tdtalbot Can you check QUERY_MODEL API call and see if we missed adding Terminal objects and if so let's add them. Let me know if you have any questions.

MaximeBaudette, 2021-09-01:
@poorva1209 
Has this been fixed in the latest release of GridApps ?

poorva1209, 2021-09-01:
@MaximeBaudette : Yes, Please give it a try and let us know if you see any more issues.

closed 2021-09-01 by poorva1209