GRIDAPPSD/gridappsd-forum #92: Problem with gridappsd installation

tehn007, 2021-02-05:
after installation by instruction (https://gridappsd.readthedocs.io/en/master/installing_gridappsd/index.html) on the ubuntu server 20.04 (address in the our test network 192.168.1.89), we cannot log in to http ://192.168.1.89: 8080
the login and password page (system manager) is displayed. 
[image]
What data do we have to enter to log in?
I tried to inject:
gridappsd, gridappsd@192.168.1.89,
Which user is needed?
 
Help me, please!
Thank you very much!

poorva1209, 2021-02-05:
After installation, have you started the platform too as stated here: https://gridappsd.readthedocs.io/en/master/using_gridappsd/index.html#start-gridapps-d-platform

poorva1209, 2021-02-05:
@tonya1 They may need help with changing the IP address in Viz. Can you provide those instructions here?

tehn007, 2021-02-05:
> After installation, have you started the platform too as stated here: https://gridappsd.readthedocs.io/en/master/using_gridappsd/index.html#start-gridapps-d-platform

Yes, of course

tonya1, 2021-02-05:
To run gridappsd on a remote system and connect with your desktop browser you will need to provide the public ip address to the viz container.  Create a viz.config file and mount it inside the viz container using the docker-compose.yml file.

On the system running gridappsd:

In the gridappsd-docker directory create a viz.config file with the following content.   Substitute the systems public ip address for localhost.

{
    "version": "remote",
    "host": "localhost:61614"
}

Edit the docker-compose.yml and add the volume information under the viz: service.

 volumes:
    - ./viz.config:/gridappsd/viz/assets/config.json

Stop and restart the platform.

./stop.sh -c
./run.sh

tehn007, 2021-02-06:
> To run gridappsd on a remote system and connect with your desktop browser you will need to provide the public ip address to the viz container. Create a viz.config file and mount it inside the viz [...]

Thank you very much for your help!
Everything works now!

poorva1209, 2021-02-08:
@tehn007 Thanks for the confirmation. I will close this issue now.

closed 2021-02-08 by poorva1209