GRIDAPPSD/gridappsd-forum #93: Document changing ip address in Viz on readthedocs

poorva1209, 2021-02-05:
Some external users are accessing Viz with remote IP address instead of localhost. If I remember correctly this needs a config file change in Viz container. Document the steps to change the config file on readthedocs so we can send that link to users.

AAndersn, 2021-07-13:
https://gridappsd-training.readthedocs.io/en/develop/installation/1.7--Lesson-1.7--Cloud-Server-Configuration.html

AAndersn, 2021-08-10:
@tonya1 or @bibiraju, please review and close

https://gridappsd-training.readthedocs.io/en/develop/installation/1.7-Cloud-Server-Configuration.html

tonya1, 2021-08-18:
@AAndersn I think we also need to add the instructions for mounting the file into the container by updating the docker-compose.yml file.

AAndersn, 2022-03-09:
Updated to use `-r` option for remote viz

https://gridappsd-training.readthedocs.io/en/develop/installation/1.7-Cloud-Server-Configuration.html

closed 2022-07-28 by poorva1209