GRIDAPPSD/gridappsd-forum #104: Running the PyVVO application

Lakes-lead, 2021-03-29:
Hello, 

I am trying to run the PyVVO application, the application logs out after a response comes up on my terminal stating that the inverter log level change to ERROR to reduce the output '

Attached is the image 

Hoping i can get some guidance  on this. 
[image]

Thank you

shpoudel, 2021-03-30:
Hello,

I am assuming your left terminal is the gridappsd-docker terminal and the right one is the application terminal. Are you getting errors in those terminals at the same time? For testing the application, you can proceed without updating the switch also (Please comment line 306 ( _update_switch_state_in_glm(glm_mgr, switches)) in app.py. This shouldn't affect the application for testing it.

Lakes-lead, 2021-03-30:
The left is the application terminal also where it was initialized and started, The right was just to view what is happening in the container during the simulation, i.e the simulation status. Both errors occurred at the same time.