GRIDAPPSD/gridappsd-forum #105: Zero values for nodes 193-51796.2 and 193-48013.3

Jeffrey-Simpson, 2021-03-30:
For nodes 193-51796.2 and 193-48013.3 I get zero measurements and very low values in the y-matrix at end up giving me divide by zero errors.

zero_nodes =  [236, 327]
The ymatrix value at these locations is (1e-12+4.14729584e-10j)

From my log:
2021-03-25 17:57:15 - main_app_new;INFO|Zero voltage value for 193-51796.2 mid _2154513b-8eb4-4458-ae4c-6bc41bfb14bc at meas id {'angle': 0.0, 'magnitude': 0.0, 'measurement_mrid': '_2154513b-8eb4-4458-ae4c-6bc41bfb14bc'}

2021-03-25 17:57:15 - main_app_new;INFO|Zero voltage value for 193-48013.3 mid _d67e7734-3e3a-4755-b51b-723a4bf99055 at meas id {'angle': 0.0, 'magnitude': 0.0, 'measurement_mrid': '_d67e7734-3e3a-4755-b51b-723a4bf99055'}

I am using the develop branch and the 9500 model.

AAndersn, 2021-03-30:
@shpoudel , these are the two phase-to-phase underground switches 

The original DSS files specify:
```
New Line.LN8979370-2         bus1=L2915534.1          bus2=193-48013.1         length=0.16924944  units=km Geometry=1P_1/0_AXNJ_DB normamps=200 emergamps=250  

New Line.WF586_48332_sw      bus1=228-979371-2_INT.3  bus2=193-48013.3         switch=True r1=0.001 r0=0.001 x1=0.001 x0=0.001 C1=0 C0=0 length=0.001 phases=1 !A-phase to C-phase

New Line.LN81048093-1        bus1=M1108393.1          bus2=193-51796.1         length=0.001797129 units=km Geometry=1P_1/0_AXNJ_DB normamps=200 emergamps=250     
                     
New Line.WD701_48332_sw     bus1=228-1048090-1_INT.2 bus2=193-51796.2         switch=True r1=0.001 r0=0.001 x1=0.001 x0=0.001 C1=0 C0=0 length=0.001 phases=1 !A-phase to B-phase

```

The VA measurement should be zero because it is an open switch, not sure why the voltage measurements are zero

temcdrm, 2021-03-30:
In OpenDSS, the switches are internal to each component. I believe you could have non-zero voltages at one (or even two) terminals, with zero (or near-zero) flow through them.

AAndersn, 2021-04-01:
The situation here is that one side of the switch is A-phase cable and the other side is a C-phase cable. The switch is used to re-assign customers on one cable to the opposite phase by cutting / opening the upstream phase conductor and then closing the switch.

Currently the model files specify the switch as not actually connecting the phases of the two cables (as inherited from the EPRI 8500 node DSS files). We could resolve the floating node issue by redefining the switch as 

```
New Line.LN8979370-2         bus1=L2915534.1          bus2=193-48013.1
New Line.WF586_48332_sw      bus1=228-979371-2_INT.3  bus2=193-48013.**1**
New Line.LN8979371-2         bus1=L2690868.3          bus2=228-979371-2_INT.3
```
This introduces a different issue that closing the switch will create a phase-to-phase fault (as it would in real-life if it were closed without doing a conductor cut first)

I'll let others chime in on how to handle this. We might get some industry input on the way these type of switches are modeled for phasing.

AAndersn, 2021-04-01:
@Jeffrey-Simpson Another idea is that you can just close the switches in the Config File API message before you start your simulation:

```
{
	"load_scaling_factor": "1",
	"schedule_name": "ieeezipload",
	"z_fraction": "0",
	"i_fraction": "1",
	"p_fraction": "0",
	"randomize_zipload_fractions": false,
	"use_houses": false,
	"model_state": { "switches": [{
				 "name": "wd701_48332_sw", "open": false},
				{"name": "wf586_48332_sw", "open": false}]
			}
}
```

shpoudel, 2021-04-01:
I don't remember exactly. But, the original version of this feeder had these switches as closed. The restoration app used to open these switches due to radiality constraints.

@AAndersn: I liked the idea of changing model_state as a temporary solution to this.

poorva1209, 2021-04-05:
Applications have to handle zero voltage.

AAndersn, 2021-04-05:
@temcdrm , I have fixed this in the DSS and CIM XML versions of the model. However, the GLM files exported by CIMHub still have the issue of the floating node:

DSS definition:
```
New Line.LN81048093-1        bus1=M1108393.1          bus2=193-51796.1         length=0.001797129 units=km Geometry=1P_1/0_AXNJ_DB normamps=200 emergamps=250                          
New Line.WD701_48332_sw      bus1=228-1048090-1_INT.2 bus2=193-51796.1         switch=True r1=0.001 r0=0.001 x1=0.001 x0=0.001 C1=0 C0=0 length=0.001 phases=1 !A-phase to B-phase
New Line.LN81048090-1        bus1=M1108433.2          bus2=228-1048090-1_INT.2 length=0.13353714  units=km Geometry=1P_1/0_AXNJ_DB normamps=200 emergamps=250 
```

CIM XML Definition:
```
<cim:SwitchPhase rdf:ID="_DC1622BC-3F79-4167-9004-0B5D9A25122D">
  <cim:IdentifiedObject.mRID>_DC1622BC-3F79-4167-9004-0B5D9A25122D</cim:IdentifiedObject.mRID>
  <cim:IdentifiedObject.name>wd701_48332_sw_B</cim:IdentifiedObject.name>
  <cim:SwitchPhase.closed>false</cim:SwitchPhase.closed>
  <cim:SwitchPhase.normalOpen>true</cim:SwitchPhase.normalOpen>
  <cim:SwitchPhase.phaseSide1 rdf:resource="http://iec.ch/TC57/CIM100#SinglePhaseKind.B"/>
  <cim:SwitchPhase.phaseSide2 rdf:resource="http://iec.ch/TC57/CIM100#SinglePhaseKind.A"/>
  <cim:SwitchPhase.Switch rdf:resource="#_63CDE28F-FFC9-4869-BC12-EC3244A056CA"/>
  <cim:PowerSystemResource.Location rdf:resource="#_259C3C83-5240-4E02-863F-29F78640A0C4"/>
</cim:SwitchPhase>
```

GLM Defintion:

```
object switch { // CIM LoadBreakSwitch
  name "swt_wd701_48332_sw";
  from "228-1048090-1_int";
  to "193-51796";
  phases B;
  status OPEN;
  continuous_rating_B 400.00;
  emergency_rating_B 600.00;
}
```

Because the GLM file isn't showing this, there still exists a floating 193-51976.B node in GridAPPS-D simulations, but not in the CIM XML. Not sure if this going to cause further trouble for apps

closed 2021-06-28 by AAndersn