GRIDAPPSD/gridappsd-forum #108: Error contracting http://localhost:8889/bigdata/ (000)

Mvr013, 2021-08-20:
I am trying to run the GRIDAPPS-D platform. 
After Getting blazegraph status I got the following error
I got error Error contracting http://localhost:8889/bigdata/ (000)

Kindly help me to run the platform.
[image]
[image]
@poorva1209 @shiva

AAndersn, 2021-08-31:
@Mvr013 Is there possibly another service currently using port 8889?

shpoudel, 2021-08-31:
I was tagged in the issue, but for some reason, I was not notified. Sorry about that.

@Mvr013 Please re-run the same command. Usually, in the first run, there is an error in blazegraph. Let me know how it goes.

AAndersn, 2021-09-20:
@Mvr013  Please let me know if you are still having this problem. Otherwise, I will close this ticket.

AAndersn, 2021-09-29:
Closing for lack of response

closed 2021-09-29 by AAndersn