GRIDAPPSD/gridappsd-forum #110: FRS messages in TimeSeries DB

MaximeBaudette, 2022-02-23:
Issue used to track advancement in implementation and integration of FRS messages into GridApps

MaximeBaudette, 2022-03-01:
@poorva1209 Could you keep me updated of the current status on that item ?

poorva1209, 2022-03-28:
@ericstephan  is currently is working on it and should have an updated API to test this week.

ericstephan, 2022-03-30:
Coding changes are complete and have been pushed to git.  @tonya1 @poorva1209 @craig8 I will create a tagged release tomorrow and send out the versions.

These changes include support for frs message type structures and advanced query interface.

ericstephan, 2022-03-30:
@poorva1209 @tonya1 Please use v1.4 tags for proven-cluster and proven-client repositories.  Stay tuned for documentation regarding some slight augmentations made to the messages to make them consistent so I could keep the user interface generic.

poorva1209, 2022-03-30:
@tonya1 Make proven container with that tag and let me know. I will give it a try then.

tonya1, 2022-03-30:
@poorva1209  Please test with docker pull gridappsd/proven:1.4.

closed 2022-05-03 by poorva1209