GRIDAPPSD/gridappsd-forum #112: Restrict Access to MySQL / REDIS

MaximeBaudette, 2022-03-23:
The cybersecurity team, where I work (LBNL), flagged my computer for exposing both MYSQL and REDIS databases to the internet. 
Is there a flag / command to execute the database containers in "local" mode where they only listen to 127.0.0.1 and dump traffic coming from somewhere else ?

MaximeBaudette, 2022-03-28:
I solved it by modifying the docker-compose by adding a more specific port binding restricting it to 127.0.0.1

```
redis:
    image: redis:3.2.11-alpine
    container_name: redis
    ports:
      - "127.0.0.1:6379:6379"

    volumes:
      - ./gridappsd/redis/data:/data
    entrypoint: redis-server --appendonly yes

  mysql:
    image: mysql/mysql-server:5.7
    container_name: mysql
    ports:
      - "127.0.0.1:3306:3306"
```

poorva1209, 2022-05-16:
@tonya1 Is this closed now ?

tonya1, 2022-05-16:
Closing this issue.

closed 2022-05-16 by tonya1