GRIDAPPSD/gridappsd-forum #118: Units in measurements

MaximeBaudette, 2022-05-09:
The simulation output gives me the following: 
```
measurements = {'_76ffe5d6-e3df-414d-aef6-02e43a2e46b9': {'angle': -84.4116663643572, 'magnitude': 564412.7477978726, 'measurement_mrid': '_76ffe5d6-e3df-414d-aef6-02e43a2e46b9'},
'_5bc15cff-98c4-45e4-bf79-65f965a728f3': {'angle': -85.0065641156567, 'magnitude': 566664.412023975, 'measurement_mrid': '_5bc15cff-98c4-45e4-bf79-65f965a728f3'},
'_29265a04-569e-465b-9ecd-7015c3288886': {'angle': -83.3488562901428, 'magnitude': 564047.0806782313, 'measurement_mrid': '_29265a04-569e-465b-9ecd-7015c3288886'}}
```

I was trying to get the total substation power by computing the following, assuming the angles were in degrees: 
```
subs_power = sum(np.real(measurements[meas_ID]['magnitude'] * np.exp(1j*np.deg2rad(measurements[meas_ID]['angle']))) for meas_ID in measurements if meas_ID in substation_measurement_IDs)
print(f"Substation power: {subs_power/1000:.3f} kW")
```

But when I look at the measurement data, it gives me mostly Q, instead of P, which I believe is wrong. 
Are the angles in radians ? Am I to subtract a reference angle ?

poorva1209, 2022-05-10:
@afisher1 @AAndersn Can you answer this?

shpoudel, 2022-05-10:
Here is my quick response to your question.

1. The angles by default are in degrees. So, you'll need to convert to radians for mathematical calculation.
2. P might be less than Q if there are DERs supplying the feeder loads. But, if the values still do not make sense, we can dig into it deeper.

MaximeBaudette, 2022-05-10:
Angles in degrees, ok! 
My issue is that in the model, the power measurements, as you can see in my post above, come back with an angle around -85 degrees. Power: 54.963 kW, Q -561.730 kVAr
That's an insane amount of Q compared to P.

AAndersn, 2022-05-11:
@MaximeBaudette  Is this for the 13OCHRE model? You can open the switches for both caps through the viz or you can add it as a difference message to the simulation start message if you are starting the simulation from Python. That will fix the power factor and Q issue you are seeing

[image]

MaximeBaudette, 2022-05-11:
@AAndersn 
Oh I see... Thanks ! 
There is however only about 200kW of load when IEEE13 is more in the ballpark of 2-4MW... so we're missing loads

MaximeBaudette, 2022-05-11:
@AAndersn 
I search here but could not locate an existing issue about the display unit for the angle
created https://github.com/GRIDAPPSD/gridappsd-forum/issues/121

closed 2022-05-11 by MaximeBaudette