GRIDAPPSD/gridappsd-forum #119: Send setpoints to PV units

MaximeBaudette, 2022-05-10:
Trying to send setpoints to PV units with the following code in python:
```
FRS_diffs = DifferenceBuilder(self.sim_id)
        for resourceID, setpoint in der_rt_control_data.items():
            if resourceID in PV_list.keys():
                FRS_diffs.add_difference(PV_list[resourceID], "PowerElectronicsConnection.p", setpoint, 0.0)
                #FRS_diffs.add_difference(resourceID, "PowerElectronicsConnection.q", setpoint, 0)
            elif resourceID in BATT_list:
                FRS_diffs.add_difference(BATT_list[resourceID], "PowerElectronicsConnection.p", setpoint, 0.0)
                #FRS_diffs.add_difference(resourceID, "PowerElectronicsConnection.q", setpoint, 0)
        self.logger.info("Dispatching der_rt_control data to the simulation engine")
        self.gapps.send(self._publish_to_topic, json.dumps(FRS_diffs.get_message())
```
Getting the following from GridApps and then the simulation crashes: 

```
2022-05-08 00:48:09,367 Thread-39 ERROR [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1651970889367|helics_goss_bridge.py|209620691|ERROR|system|ERROR
Error in run simulation Traceback (most recent call last):
  File "/gridappsd/services/helicsgossbridge/service/helics_goss_bridge.py", line 748, in _publish_to_helics_bus
    raise RuntimeError(f"Forward difference command cannot be parsed correctly one or more of attributes needed was None.\ndifference:{json.dumps(x,indent=4,sort_keys=True)}\nparsed result:{json.dumps(parsed_result,indent=4,sort_keys=True)}")
RuntimeError: Forward difference command cannot be parsed correctly one or more of attributes needed was None.
difference:{
    "attribute": "PowerElectronicsConnection.p",
    "object": "_B1C5C0AF-EBE8-4D99-894A-C93D037F9A26",
    "value": 0.023424722997332587
}
parsed result:{
    "cim_attribute": "PowerElectronicsConnection.p",
    "object_name": null,
    "object_name_prefix": null,
    "object_phases": null,
    "object_property_list": null,
    "object_total_phases": null,
    "object_type": null
}

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/gridappsd/services/helicsgossbridge/service/helics_goss_bridge.py", line 552, in run_simulation
    self._publish_to_helics_bus(self._simulation_command_queue.get(), self._command_filter)
  File "/gridappsd/services/helicsgossbridge/service/helics_goss_bridge.py", line 901, in _publish_to_helics_bus
    raise RuntimeError(err_msg)
RuntimeError: An error occured while trying to translate the update message received
Traceback (most recent call last):
  File "/gridappsd/services/helicsgossbridge/service/helics_goss_bridge.py", line 748, in _publish_to_helics_bus
    raise RuntimeError(f"Forward difference command cannot be parsed correctly one or more of attributes needed was None.\ndifference:{json.dumps(x,indent=4,sort_keys=True)}\nparsed result:{json.dumps(parsed_result,indent=4,sort_keys=True)}")
RuntimeError: Forward difference command cannot be parsed correctly one or more of attributes needed was None.
difference:{
    "attribute": "PowerElectronicsConnection.p",
    "object": "_B1C5C0AF-EBE8-4D99-894A-C93D037F9A26",
    "value": 0.023424722997332587
}
parsed result:{
    "cim_attribute": "PowerElectronicsConnection.p",
    "object_name": null,
    "object_name_prefix": null,
    "object_phases": null,
    "object_property_list": null,
    "object_total_phases": null,
    "object_type": null
}

2022-05-08 00:48:09,401 Thread-39 ERROR [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1651970889397|gov.pnnl.goss.gridappsd.simulation.SimulationProcess$1|Simulator-209620691|ERROR|system|ERROR
ERROR    [2022-04-01 20:38:50 UTC] : received SIGABRT; exiting now

```

poorva1209, 2022-05-10:
@afisher1 Looks like helics bridge is throwing an error. Can you check whats causing this error?

shpoudel, 2022-05-10:
This is not aa HELICS error. This is an issue with incorrect object MRID.

Maxime, you're using pecid to control the PV unit. Please see the email that I sent on 04/29. Please use id, not the pecid and the message should go without an error. 

The **pv_mrid** and **batt_mrid** can be obtained from the query in the email. Please use "**id**" parameters from the query response to control the PV/Batt.

MaximeBaudette, 2022-05-10:
Thanks Shiva,
I will try that!

AAndersn, 2022-05-10:
Queries are also included in https://github.com/GRIDAPPSD/gridappsd-training/issues/7

The queries for the device mRIDs should be passed using the PowerGrid Models API: https://gridappsd-training.readthedocs.io/en/latest/api_usage/3.3-Using-the-PowerGrid-Models-API.html#Query-using-a-Generic-SPARQL-Query

closed 2022-05-11 by MaximeBaudette

shpoudel, 2022-05-11:
> Check twice before you click! This email originated from outside PNNL.
>
> @shpoudel<https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fshpoudel&data=05%7C01%7Cshiva.poudel%40pnnl.gov%7Cf48774143ee643767bfb08da336e0684%7Cd6faa5f90ae240338c0130048a38deeb%7C0%7C0%7C637878844890848051%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=IvKpF7rIGFKZwfx14KDPjVH05wc%2B4CZPX0F7rg84jAI%3D&reserved=0>
> On our original call you mentioned I should look for POwerElectroniConnections cim object, so I used these IDs originally. With the answer from @AAndersn<https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FAAndersn&data=05%7C01%7Cshiva.poudel%40pnnl.gov%7Cf48774143ee643767bfb08da336e0684%7Cd6faa5f90ae240338c0130048a38deeb%7C0%7C0%7C637878844890848051%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=jzh2mIu7q7lKyYzZs%2BRktYgrEKWGuzphY7Rv%2BgFSYsU%3D&reserved=0> I see that I should have looked up the PhotovoltaicUnits cim object instead.
> I nmo longer have the error now that I did that, thanks for the help !

See this if you're interested. I specifically mentioned over my email to use the query and "id", not the "pecid". What Alexander answered is the easy way to get PV ids. At the end of the day, they are all related to powerelectronicsconnection.

[cid:4c3d276e-34ab-4b36-b46e-80eb84a12252]

MaximeBaudette, 2022-07-11:
Still having some issues regarding building difference for my simulation. 
I tried to update the mRID to the ones of the PowerElectronicsConnection, but it not working either. 

I'm very confused by what you call ID vs PECID, considering that neither are fields in the CIM model... 

I tried to run your query, but without success either

```
from gridappsd import topics as t
import os
topic = t.REQUEST_POWERGRID_DATA

# Re-authenticate with GridAPPS-D (the java security token doesn't get inherited well)
os.environ['GRIDAPPSD_USER'] = 'app_user'
os.environ['GRIDAPPSD_PASSWORD'] = '1234App'
try:
    gapps = GridAPPSD()
    assert gapps.connected # Connect to GridAPPS-D Platform
except:
     gapps = None

query_string = 'PREFIX r: <http://www.w3.org/1999/02/22-rdf-syntax-ns#> PREFIX c: <http://iec.ch/TC57/CIM100#> PREFIX xsd: <http://www.w3.org/2001/XMLSchema#> SELECT ?name ?bus ?ratedS ?ratedU ?ipu ?p ?q ?fdrid ?id ?pecid (group_concat(distinct ?phs;separator="\n") as ?phases) WHERE {VALUES ?feeder_mrid { "_C07972A7-600D-4AA5-B254-4CAA4263560E"} ?s r:type c:PhotovoltaicUnit. ?s c:IdentifiedObject.name ?name. ?s c:IdentifiedObject.mRID ?id. ?pec c:PowerElectronicsConnection.PowerElectronicsUnit ?s. ?pec c:IdentifiedObject.mRID ?pecid. ?pec c:Equipment.EquipmentContainer ?fdr. ?fdr c:IdentifiedObject.mRID ?feeder_mrid. ?pec c:PowerElectronicsConnection.ratedS ?ratedS. ?pec c:PowerElectronicsConnection.ratedU ?ratedU. ?pec c:PowerElectronicsConnection.maxIFault ?ipu. ?pec c:PowerElectronicsConnection.p ?p. ?pec c:PowerElectronicsConnection.q ?q. OPTIONAL {?pecp c:PowerElectronicsConnectionPhase.PowerElectronicsConnection ?pec. ?pecp c:PowerElectronicsConnectionPhase.phase ?phsraw. bind(strafter(str(?phsraw),"SinglePhaseKind.") as ?phs) } ?t c:Terminal.ConductingEquipment ?pec. ?t c:Terminal.ConnectivityNode ?cn. ?cn c:IdentifiedObject.name ?bus} GROUP by ?name ?bus ?ratedS ?ratedU ?ipu ?p ?q ?fdrid ?id ?pecid ORDER by ?name'
message = {
    "requestType": "QUERY",
    "queryString": query_string,
    "resultFormat": "JSON"
 }

gapps.get_response(topic, message, timeout=1000)

```

I get the following error back:

```
{'error': {'message': 'org.apache.jena.query.QueryParseException: Lexical error at line 1, column 487.  Encountered: "\\n" (10), after : "\\""\n\tat org.apache.jena.sparql.lang.ParserSPARQL11.perform(ParserSPARQL11.java:109)\n\tat org.apache.jena.sparql.lang.ParserSPARQL11.parse$(ParserSPARQL11.java:52)\n\tat org.apache.jena.sparql.lang.SPARQLParser.parse(SPARQLParser.java:34)\n\tat org.apache.jena.query.QueryFactory.parse(QueryFactory.java:147)\n\tat org.apache.jena.query.QueryFactory.create(QueryFactory.java:79)\n\tat org.apache.jena.query.QueryFactory.create(QueryFactory.java:52)\n\tat org.apache.jena.query.QueryFactory.create(QueryFactory.java:40)\n\tat gov.pnnl.goss.gridappsd.data.handlers.BlazegraphQueryHandler.query(BlazegraphQueryHandler.java:96)\n\tat gov.pnnl.goss.gridappsd.data.BGPowergridModelDataManagerImpl.queryResultSet(BGPowergridModelDataManagerImpl.java:226)\n\tat gov.pnnl.goss.gridappsd.data.BGPowergridModelDataManagerImpl.query(BGPowergridModelDataManagerImpl.java:201)\n\tat gov.pnnl.goss.gridappsd.data.BGPowergridModelDataManagerHandlerImpl.handle(BGPowergridModelDataManagerHandlerImpl.java:32)\n\tat gov.pnnl.goss.gridappsd.data.DataManagerImpl.processDataRequest(DataManagerImpl.java:149)\n\tat gov.pnnl.goss.gridappsd.process.ProcessEvent.onMessage(ProcessEvent.java:264)\n\tat pnnl.goss.core.client.GossClient$1.run(GossClient.java:398)\n\tat java.lang.Thread.run(Thread.java:748)\n'},
 'responseComplete': True,
 'id': '6df8c421-9841-4b38-bde6-fe0777537c4c'}
```

AAndersn, 2022-07-13:
@MaximeBaudette - This an issue with how python interprets wrapped quotes and the line break characters. Try deleting this part of the query and it should work:

`(group_concat(distinct ?phs;separator="\n") as ?phases)`

There is a special punctuation fix that I worked out some time ago. I think it was wrapping the character like `'"\\n"'`, but I need to go dig it out of my old code.

MaximeBaudette, 2022-07-13:
Thanks, I forgot to mention but Shiva helped me investigate the issue