GRIDAPPSD/gridappsd-forum #120: Load updated IEEE13OCHRE in Blazegraph

AAndersn, 2022-05-11:
@tonya1, Please create a new Blazegraph container with an updated version of the IEEEOCHRE model

@vmotakat please email Tonya the new CIM XML and measurements.txt files generated by the CIMHub import script.

AAndersn, 2022-05-11:
@tonya1 When you have built the container, please share the tag with @MaximeBaudette

MaximeBaudette, 2022-05-11:
In the meantime, I seem to be having issues launching a OCHRE simulation... 
Let me track that in a separate issue over here https://github.com/GRIDAPPSD/gridappsd-forum/issues/122

tonya1, 2022-05-11:
@MaximeBaudette The test container has been shared through the collaboration site.  You should be receiving an email with instructions to download.

MaximeBaudette, 2022-05-11:
Excellent, I just started downloading it !
Thanks !

MaximeBaudette, 2022-05-11:
@tonya1 
Assuming the new tag is `ieee13_ocher_der` and no longer `fastderms` ?

tonya1, 2022-05-11:
@MaximeBaudette  you are correct .

closed 2022-05-16 by AAndersn