GRIDAPPSD/gridappsd-forum #121: Wrong display unit for angles on Gridapps-D web GUI

MaximeBaudette, 2022-05-11:
The display unit on the gridapps-D web GUI are wrong and use the quantity base unit (V, W), should display degrees instead
See screenshot
[image]

shpoudel, 2022-05-11:
@bibiraju To fix this, look for the **check box** in "Component type". If the user checks the angle, then display "Degrees" as Y-label in the plot.

bibiraju, 2022-05-12:
So for angle measurements, the unit should be degrees like the picture below?

[image]

bibiraju, 2022-05-12:
@shpoudel should the unit for Power magnitude be VA?

shpoudel, 2022-05-12:
It is VA, but let's bring this to a standup call or a working meeting if we need to change this to W or Var with some additional math.

closed 2022-05-26 by poorva1209