GRIDAPPSD/gridappsd-forum #122: Not able to start ochre simulation

MaximeBaudette, 2022-05-11:
I'm trying to run a ochre model, as per our discussion on the call this morning, I changed the "simulator" in my simulation configuration to "OCHRE", but it doesn't seem to work. 

```
2022-05-11 21:22:02,020 Thread-39 INFO [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1652304122013|gov.pnnl.goss.gridappsd.simulation.SimulationManagerImpl|2091347531|STARTING|system|INFO
Starting simulation 2091347531

2022-05-11 21:22:02,030 Thread-39 INFO [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1652304122022|gov.pnnl.goss.gridappsd.process.ProcessNewSimulationRequest|2091347531|RUNNING|system|INFO
Started simulation for id 2091347531

2022-05-11 21:22:02,046 Thread-39 INFO [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1652304122031|gov.pnnl.goss.gridappsd.simulation.SimulationProcess|2091347531|RUNNING|system|INFO
Setting num federates 

2022-05-11 21:22:02,070 Thread-76 ERROR [gov.pnnl.goss.gridappsd.simulation.SimulationProcess] - Error during simulation
java.io.IOException: Cannot run program "helics" (in directory "/tmp/gridappsd_tmp/2091347531"): error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at gov.pnnl.goss.gridappsd.simulation.SimulationProcess.run(SimulationProcess.java:177)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 1 more
2022-05-11 21:22:02,103 Thread-39 INFO [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1652304122045|gov.pnnl.goss.gridappsd.simulation.SimulationProcess|2091347531|RUNNING|system|INFO
Starting gridlabd simulator with command gridlabd.sh /tmp/gridappsd_tmp/2091347531/model_startup.glm

2022-05-11 21:22:02,122 Thread-39 INFO [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1652304122055|gov.pnnl.goss.gridappsd.simulation.SimulationProcess|2091347531|RUNNING|system|INFO
Command for ochre ready helics run --path /tmp/gridappsd_tmp/2091347531/ochre_helics_config.json --bro
2022-05-11 21:22:02,147 Thread-39 INFO [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1652304122058|gov.pnnl.goss.gridappsd.simulation.SimulationProcess|2091347531|RUNNING|system|INFO
Starting simulator with command helics run --path /tmp/gridappsd_tmp/2091347531/ochre_helics_config.js
2022-05-11 21:22:02,163 Thread-39 ERROR [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1652304122072|gov.pnnl.goss.gridappsd.simulation.SimulationProcess|2091347531|ERROR|system|ERROR
Simulation error: Cannot run program "helics" (in directory "/tmp/gridappsd_tmp/2091347531"): error=2, No such file or directory

2022-05-11 21:22:11,828 Thread-39 INFO [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1652304131820|gov.pnnl.goss.gridappsd.process.ProcessEvent|2073737646|RUNNING|system|INFO
New process id generated with new process type

2022-05-11 21:23:05,586 Thread-39 INFO [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1652304185576|gov.pnnl.goss.gridappsd.process.ProcessEvent|614809666|RUNNING|system|INFO
New process id generated with new process type

2022-05-11 21:23:05,747 Thread-46 ERROR [gov.pnnl.proven.api.exchange.RestExchange] - 
Error while calling REST Service
2022-05-11 21:23:05,747 Thread-46 ERROR [gov.pnnl.proven.api.exchange.RestExchange] - Exception: javax.ws.rs.InternalServerErrorException: HTTP 500 Internal Server Error
2022-05-11 21:23:05,779 Thread-39 ERROR [gov.pnnl.goss.gridappsd.log.LogManagerImpl] - 1652304185748|gov.pnnl.goss.gridappsd.process.ProcessEvent|614809666|ERROR|system|ERROR
javax.ws.rs.InternalServerErrorException: HTTP 500 Internal Server Error
	at org.glassfish.jersey.client.JerseyInvocation.convertToException(JerseyInvocation.java:1098)
	at org.glassfish.jersey.client.JerseyInvocation.translate(JerseyInvocation.java:883)
	at org.glassfish.jersey.client.JerseyInvocation.lambda$invoke$1(JerseyInvocation.java:767)
	at org.glassfish.jersey.internal.Errors.process(Errors.java:316)
	at org.glassfish.jersey.internal.Errors.process(Errors.java:298)
	at org.glassfish.jersey.internal.Errors.process(Errors.java:229)
	at org.glassfish.jersey.process.internal.RequestScope.runInScope(RequestScope.java:414)
	at org.glassfish.jersey.client.JerseyInvocation.invoke(JerseyInvocation.java:765)
	at org.glassfish.jersey.client.JerseyInvocation$Builder.method(JerseyInvocation.java:456)
	at org.glassfish.jersey.client.JerseyInvocation$Builder.post(JerseyInvocation.java:357)
	at gov.pnnl.proven.api.exchange.RestExchange.addProvenData(RestExchange.java:341)
	at gov.pnnl.proven.api.producer.ProvenSession.sendMessage(ProvenSession.java:238)
	at gov.pnnl.proven.api.producer.Producer.sendMessage(Producer.java:415)
	at gov.pnnl.proven.api.producer.ProvenProducer.getAdvancedTsQuery(ProvenProducer.java:139)
	at gov.pnnl.goss.gridappsd.data.ProvenTimeSeriesDataManagerImpl.query(ProvenTimeSeriesDataManagerImpl.java:150)
	at gov.pnnl.goss.gridappsd.data.ProvenTimeSeriesDataManagerImpl.handle(ProvenTimeSeriesDataManagerImpl.java:136)
	at gov.pnnl.goss.gridappsd.data.DataManagerImpl.processDataRequest(DataManagerImpl.java:149)
	at gov.pnnl.goss.gridappsd.process.ProcessEvent.onMessage(ProcessEvent.java:264)
	at pnnl.goss.core.client.GossClient$1.run(GossClient.java:398)
	at java.lang.Thread.run(Thread.java:748)
```

AAndersn, 2022-05-11:
@MaximeBaudette I think FAST-DERMS may be using a different OCHRE version than the default version in the GridAPPS-D platform

@vmotakat, please share with Maxime the instructions for adding latest OCHRE into the docker-compose.yml file

MaximeBaudette, 2022-05-11:
@AAndersn 
I'll see with @vmotakat for a temporary fix for now. 
Could you discuss that  maybe it would be pertinent to setup a dedicated FASTDERMS branch on GridApps-D so that we have the needed model / data / functionalities and services for the project. 
The whole point of having a platform is to be able to pull it and run it without having to fiddle with custom installs ?

MaximeBaudette, 2022-05-11:
@AAndersn @poorva1209 
I had a chat with Venkat and we were not able to make OCHRE working on my computer, he mentioned something related to GOSS.
Could you update gridapps with the necessary components to be able to run OCHRE simulations ? and direct me to the relevant tag ?

AAndersn, 2022-05-12:
@MaximeBaudette Are you running the latest version of GridAPPS-D? 

The latest ~develop or any of the 2022 releases should have everything you need to run OCHRE. To make sure, do a `git pull` from your gridappsd-docker directory and then do `./run.sh -t develop` or `./run.sh -t releases_2022.04.0`

MaximeBaudette, 2022-05-12:
I have been running `./run.sh -t v2022.04.0` for a while now...

AAndersn, 2022-05-12:
> @AAndersn I'll see with @vmotakat for a temporary fix for now. Could you discuss that maybe it would be pertinent to setup a dedicated FASTDERMS branch on GridApps-D so that we have the needed model [...]

@MaximeBaudette When I spoke with @tonya1 this afternoon, she said that we had been specifically told not to push the FAST-DERMS materials to the public github repository. We would need a public version of the ochre simulator, etc. The version that Venkat gave you is only on the NREL private repository and not accessible even by me.

MaximeBaudette, 2022-05-12:
So I'm really confused, do I need a custom install, or does it come standard ?

AAndersn, 2022-05-12:
> I have been running `./run.sh -t v2022.04.0` for a while now...

Ok, you might try downgrading to releases_2022.01.0. I have done my integration testing with Venkat and the SDGE feeder and know for sure that everything worked fine for that release

AAndersn, 2022-05-12:
> So I'm really confused, do I need a custom install, or does it come standard ?

There is an OCHRE version that comes standard and is from last year's build.

However, Harsha, Jason, and others have made a lot of changes to their own OCHRE setup, but have not shared any of those changes with us

AAndersn, 2022-05-12:
@MaximeBaudette, Venkat and I have been able to verify that a HELICS co-simulation bridge initialization failure error will be thrown if the "/NREL-Battery-Aggrigator/gridappsd-ochre/inputs/MS/Main_spreadsheet_FASTDERM.xlsx" spreadsheet is missing or uses a different name from that specified in the gridappsd-ochre/constants.py file.

This error goes away as soon as spreadsheet is put back into the correct directory. Running the simulation from Viz or through the python API makes no difference.

AAndersn, 2022-05-16:
It was determined this was a hardware resource issue with insufficient RAM to run docker on new MAC devices. Related to #123

closed 2022-05-16 by AAndersn

AAndersn, 2022-10-11:
Maxime,

When I spoke with Tonya this afternoon, she said that we had been
specifically told not to push that particular version of FAST-DERMS
materials to the public github repository. We would need a public version
of the ochre simulator, etc. The version that Venkat gave you is only on
the NREL private repository and not accessible even by me.