GRIDAPPSD/gridappsd-forum #125: LBL FastDERMS unable to parse modified 13ochre using PyCIM

AAndersn, 2022-05-16:
@MaximeBaudette @jsmacdonald
```
Traceback (most recent call last):
  File "/home/ubuntu/FRS_SIM/fastderms/src/FAST_DERMS_CONTROL/IOs/IOclass.py", line 50, in fetch_model
    import_result = cim.cimread(source)
  File "/home/ubuntu/miniconda3/lib/python3.8/site-packages/PyCIM/RDFXMLReader.py", line 61, in cimread
    namespaces = xmlns(source)
  File "/home/ubuntu/miniconda3/lib/python3.8/site-packages/PyCIM/RDFXMLReader.py", line 233, in xmlns
    for (event, elem) in iterparse(source, events):
  File "/home/ubuntu/miniconda3/lib/python3.8/xml/etree/ElementTree.py", line 1227, in iterator
    yield from pullparser.read_events()
  File "/home/ubuntu/miniconda3/lib/python3.8/xml/etree/ElementTree.py", line 1302, in read_events
    raise event
  File "/home/ubuntu/miniconda3/lib/python3.8/xml/etree/ElementTree.py", line 1274, in feed
ERROR:__Main__Orchastrator:Orchastrator failed
    self._parser.feed(data)
xml.etree.ElementTree.ParseError: not well-formed (invalid token): line 1, column 16
```

MaximeBaudette, 2022-05-17:
@AAndersn 
Did you install the dependencies from pipenv ? 
I setup the development environment using pipenv: https://pipenv.pypa.io/en/latest/
It requires python 3.7 and I see error messages about python 3.8, could you change the kernel for python 3.7 ?

MaximeBaudette, 2022-05-17:
talked to Jason, and mentioned that since we had conda, and you deployed miniconda on the server, there's prob a bunch of base packages that were already installed on all our machines that we forgot to add to the pipenv file. He's gonna investigate today and try to fix the kernel install managed through pipenv
.

AAndersn, 2022-05-23:
PyCIM appears to be able to parse the file obtained directly from the PowerGrid Models API, so closing this issue for now. Will re-open if the issue continues.

closed 2022-05-23 by AAndersn