GRIDAPPSD/gridappsd-forum #150: XML to GridAPPS-D file conversion issue for regulators

vmotakat, 2024-07-01:
Hi,

I am having issues when I am converting XML files to GridAPPS-D files of feeders having regulators. I noticed regulator information in the XML file but not in either the GML file, DSS file, or conversion_dict.json file.

Can anyone help me find the issue in CIMHUB python codes of GridAPPS-D?

shpoudel, 2024-07-08:
@vmotakat : Can you share with me the following information?
1. OpenDSS version you are using (opendsscmd probably).
2. CIMHub branch that you are using.

poorva1209, 2024-07-08:
> @vmotakat : Can you share with me the following information?
> 
> 1. OpenDSS version you are using (opendsscmd probably).
> 2. CIMHub branch that you are using.

As per Venkat's reply to me: 
CIMHub branch: * final9500
OpenDSSdirect

shpoudel, 2024-07-09:
OpenDSS direct is a Python interface to OpenDSS. 

@vmotakat: Please use version 1.2.17 of OpenDSS, and the "develop" branch of CIMHub. They should be compatible with each other.